hishizuka/pizero_bikecomputer #1: real time W'bal integration as info displayed

serpico, 2021-07-20:
It's more of an idea than feature request ( I'll probably never build a bike computer even given a very detailled how to like you're doing here  :-) ). 

I, nevertheless, appreciate your great work. 

I'm using using [Goldencheetah](https://github.com/GoldenCheetah/GoldenCheetah) to keep track of my bike rides and I thought it would be great for some bike tourers to have that W' bal info ( the red curve on the screenshot below ) directly available on the screen, in ( almost, recalculated every few minutes, kilometers... ) real time to, for example not overwork themselves during sorties. I didn't real dig on the what that would involve but at first glance, it seems you're already packing all what's needed on your board. 

Again it's just an idea...and congratulation 👍 on what you did.

[image]

hishizuka, 2021-09-07:
Hi, thanks for your suggestion.
It looks like it can be done with the following repository that Goldencheetah team is working on.
https://github.com/GoldenCheetah/sweatpy

hishizuka, 2022-01-13:
Hi, @serpico, I've finally implemented it.
The input parameters are CP and W prime, and the W prime balance is updated in real time.
I' ll update the repository soon.

[image]

hishizuka, 2022-01-23:
Here are some screenshots.

[image]

[image]

closed 2022-03-07 by hishizuka