hishizuka/pizero_bikecomputer #2: AttributeError: module 'PyQt5.QtGui' has no attribute 'QTextEdit'

slumsdon, 2021-09-07:
Hi Hishi
Great project and I am enjoying trying to get this to work. Diligently working my way through the hardware and software installs and have a timeout issue with ANT+ and some questions about using the waveform 2.4" display

But main issue is trying to get the graphics to work in console mode, keep getting this issue

Traceback (most recent call last):
  File "pizero_bikecomputer.py", line 49, in <module>
    main()
  File "pizero_bikecomputer.py", line 39, in main
    gui = gui_pyqt.GUI_PyQt(config)
  File "/home/pi/pizero_bikecomputer/modules/gui_pyqt.py", line 108, in __init__
    self.init_window()
  File "/home/pi/pizero_bikecomputer/modules/gui_pyqt.py", line 187, in init_window
    self.debug_log_viewer_widget = DebugLogViewerWidget(self.stack_widget, "Debug Log Viewer", self.config)
  File "/home/pi/pizero_bikecomputer/modules/pyqt/menu/pyqt_menu_widget.py", line 37, in __init__
    self.setup_ui()
  File "/home/pi/pizero_bikecomputer/modules/pyqt/menu/pyqt_menu_widget.py", line 65, in setup_ui
    self.setup_menu()
  File "/home/pi/pizero_bikecomputer/modules/pyqt/menu/pyqt_debug_widget.py", line 32, in setup_menu
    self.editor = QtWidgets.QTextEdit() if USE_PYQT6 else QtGui.QTextEdit()
AttributeError: module 'PyQt5.QtGui' has no attribute 'QTextEdit'

Any help will be much appreciated
THAnks
Simon

hishizuka, 2021-09-08:
Hi, thanks for using my work.
I fixed the bug, please try it.

slumsdon, 2021-09-08:
Fantastic, many thanks, yes that fixed it! Graphics now working. Now need to fix USB timeout error, missing fb1 framebuffer and GPS data missing. Pressure data is working so thats something

THanks again
Simon

hishizuka, 2021-09-08:
Hello, I will support to work.

ANT+ USB timeout error
This is an error message by openant, and my program does not handle it specifically. If the ANT+ sensor is paired, you will not get the error message. Do you have any ANT+ sensors?　If you don't have any, then I suggest you uninstall openant itself.

Missing fb1 framebuffer 

First, which LCD are you using? I've only used Adafruit's PiTFT.
Follow the seller's tutorial and make sure it draws on /dev/fb1.
Here is a tutorial I wrote on PiTFT in Japanese.
https://qiita.com/hishi/items/bdd630666277e4f8162a

Pressure data
Which product do you use for the barometric altimeter?

slumsdon, 2021-09-08:
Hi Hishi

Many thanks for your help

This is the display https://thepihut.com/products/2-4-lcd-display-module-240x320 from waveshare, i can get it to mirror the hdmi output on fb0 which is a good start but havent worked out how to create an fb1 framebuffer for its own output. I will study

The GPS & pressure is coming from the Ozzmaker board https://thepihut.com/products/ozzmaker-berrygps-imu-v3 and i am using the qwiic connector lead. GPS is over i2c and has problems which i need to study. Pressure is working

ANT+ also has problems connecting, i need this for HRM and hopefully power

I also have a UPS board https://thepihut.com/products/ups-hat-for-raspberry-pi-zero which works but has a connection error

I also need to wite the button shim in

Pizero_bikecomputer now starts up and looks to be working correctly which is great, it just needs gps and ant+ and then the real work can begin. There is a lot going on so i need to be methodical and work the problems one by one. But good progress is being made and i am looking forwars to seeing this working, its a nice project

When i have more useful feedback i will send detail and hopefully some photos of the setup and case.

Thanks

Simon

https://qiita.com/hishi/items/bdd630666277e4f8162a<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fqiita.com%2Fhishi%2Fitems%2Fbdd630666277e4f8162a&data=04%7C01%7C%7C5017c9f2c525468456c408d97317c57d%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637667368193166648%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=E8imQzG8wWwFcF2VJYLFIR2oj0NQjGno1p4%2F2KnabUM%3D&reserved=0>

hishizuka, 2021-09-08:
Hi Simon

Thank you for working on this project!

Display
A temporary solution is to set the resolution to 320x240 in /boot/config.txt, output to /dev/fb0 (HDMI) via linuxfb, and mirror the HDMI output to TFT.

BerryGPS IMU v4
Sorry, I have two BerryGPS IMU v4 boards and both have the same problem.
The first is that the GPS chip antenna doesn't seem to work properly and I need to use a powerful external antenna to get the GPS to work.
The second is that I haven't tried an I2C connection for this GPS, and there doesn't seem to be a suitable python pip package. If possible, use HAT connection (UART+I2C) to get position data from GPSd.
Thirdly, the magnetometer sensor does not work properly. Please give up on the electronic compass. The accelerometer is working, so the auto start/stop with vibration detection works. This works well with GPS tracking.

ANT+ sensor
If you start by checking the connection of the heart rate monitor with Openant's example program, it will be easier to isolate the problem.

If you have any problems, please give your feedback!