hishizuka/pizero_bikecomputer #4: Need some help with I2C sensor configuration

inlovewithsilver, 2022-04-01:
Hi, hishi, hope you are doing well. I'd like to firstly appreciate your work, but something going wrong with my I2C sensor - BMP280. For some reason I couldn't buy the product like _Adafruit_ and _SparkFun Qwiic SHIM for Raspberry Pi (https://www.sparkfun.com/products/15794)_, so I use _GY-BMP280-3.3_ as substitution. While I connecting the _VCC_, _GND_, _SCL_ and _SDA_ pins with Raspberry Pi (and _CSB_, _SDO_ are left vacant), bikecomputer seems not have receive the data form _BMP280_, and the demo mode shows:

> Loading original modules...
>         config : 2.914984 sec
>   GPS :  True
>   GPS_ADAFRUIT_UART :  False
>   GPS_I2C :  False
>   ANT :  False
>   RPi GPIO :  True
>   I2C :  True
> ...
>   detected I2c sensors:
>   PiTFT 2.8(r) :  True

Is my connection wrong or the configuration is not completed? Looking forward to your solution, thanks again.

hishizuka, 2022-04-02:
Hi @inlovewithsilver , thank you for using my program!

How about the result of "$ i2cdetect -y 1"?
I think the address of your sensor is differnt from the address(0x77) which I write in the program.
My result as follows,
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- 1d -- --
20: -- -- -- -- -- -- -- -- -- 29 -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 3f
40: -- -- -- -- -- -- -- -- -- 49 -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- 77

Probably, yours is 0x76.
If so, you need to modify modules/sensor/i2c/BMP280.py.
The address is specified in the line 62.

  original: SENSOR_ADDRESS = 0x77
  modified: SENSOR_ADDRESS = 0x76

Sincerely,

inlovewithsilver, 2022-04-02:
It works! The problem is just as you said, and thank you for giving me the solution. You are awesome!

hishizuka, 2022-04-02:
@inlovewithsilver . Glad it worked! Enjoy!

closed 2022-04-02 by hishizuka