hishizuka/pizero_bikecomputer #7: Question Regarding Compatibility

Seuong, 2022-10-10:
Currently there seems to be a shortage of Raspberry Pi Zeros, as a result I was wondering if I could use a Radxa Zero to the same effect.  If so, what would be different with hardware & software? If you're not sure, no worries.

hishizuka, 2022-10-10:
Hello,
I have already checked the compatibility of Radxa Zero, and I have confirmed that the required interfaces (I2C, SPI, UART, PWM) are satisfied. I posted this article.
https://forum.radxa.com/t/guide-sparkfun-qwiic-adafruit-stemma-qt/8149

Note that when using the MIP display, the PWM pins have different pins than the Raspberry Pi. If you do not need a backlight, do not connect it.

A future release will support Radxa Zero.
Stay tuned!

P.S.
Support for Mango Pi will be a long time away, because I don't know how to use the interface.

Seuong, 2022-10-10:
Awesome, thanks for the confirmation, I look forward to it!

closed 2022-10-10 by Seuong