hishizuka/pizero_bikecomputer #8: Compatibility with Raspberry Pi Pico W

MauriceArikoglu, 2022-10-31:
Hi there,

can you confirm the project working on a RPi Pico W MCU?
If not, what is required compared to the RPi Zero SBC, that the Pico doesn't offer?

Best & thanks in advance

hishizuka, 2022-11-01:
Hi,

My project needs pure python and libraries.
See [the software doc](https://github.com/hishizuka/pizero_bikecomputer/blob/master/doc/software_installation.md).

MauriceArikoglu, 2022-11-01:
@hishizuka Hi Hishi, thanks. The Pico supports MicroPython, a full python3 implementation.
I will research whether the libraries in question are supported.

hishizuka, 2022-11-01:
Hi,

Micropython is not pure Python.

The following commands at the beginning of [the software documentation](https://github.com/hishizuka/pizero_bikecomputer/blob/master/doc/software_installation.md),

`pip3 install PyQt5 numpy oyaml pillow polyline aiohttp`

PyQt and pillow definitely do not exist in MicroPython and I don't know how far ulab can be used in place of numpy.

I don't have a Pico anyway, so I have no intention of implementing it, but it would probably require re-implementing a significant portion of the program.

closed 2022-11-01 by hishizuka