hishizuka/pizero_bikecomputer #16: Start/stop button not working

Ptosiek, 2023-08-27:
Hi,

Thanks for the project. I've been playing with it for the last couple days and have started to change a few things/fix what I see.
Are you even open/do you have time for PR ?

First one would be that the start/stop button is broken, getting:
Traceback (most recent call last):
File "/home/antonin/Work/pizero_bikecomputer/modules/pyqt/pyqt_button_box_widget.py", line 112, in gui_start_and_stop_quit
self.config.logger.start_and_stop_manual()
File "/home/antonin/Work/pizero_bikecomputer/modules/logger_core.py", line 292, in start_and_stop_manual
self.config.gui.change_start_stop_button(self.config.G_MANUAL_STATUS)
File "/home/antonin/Work/pizero_bikecomputer/modules/gui_pyqt.py", line 562, in change_start_stop_button
self.button_box_widget.change_start_stop_button(status)
File "/home/antonin/Work/pizero_bikecomputer/modules/pyqt/pyqt_button_box_widget.py", line 115, in change_start_stop_button
icon = QtGui.QIcon(self.config.gui.icon_dir+'img/pause_white.png')
AttributeError: 'GUI_PyQt' object has no attribute 'icon_dir'

on click.

As a side note, I find disconcerting the icons. I would expect them to be switched between the state, eg they represent the state we wants to set, not the one we are in. So next_white.png is displayed when we are in stop/pause state whereas pause_white.png is displayed when we are in start state.

closed 2023-08-28 by hishizuka

hishizuka, 2023-08-28:
Hi, @Ptosiek ,
thanks for the pull request and suggested fixes. I have incorporated them into the source code.
Try it out!

Ptosiek, 2023-08-29:
Great!

I still have some issues, some I fixed, others still in the process of debugging.  
I've brought some substantial changes and don't want to burden your project if you don't agree with them, so not opening issues. But I've done the following in my fork, if you wanna have a look.
There's a lot of branches, so we could in theory cherry-pick the changes. 

(between () is the branch name)

1. (pip-tools) Add dependency management with requirements files/use of pip-tools to get a lock file. Couldn't get poetry to work because of Cython.

It handles only software requirements, but not yet the libs to install for hardware (to come in later branches). 

Btw, I've also forked crdp to have the install working properly (even with cython no being previously installed). 
This could go in any case: https://github.com/Ptosiek/crdp/commit/c82305c0a242a7fc4230bcb8027166e35868edfd

2. (add-black) Use black to format code.

I haven't added any automated way of formatting the code (through actions or pre-commit) because I don't want to change the flow too much if you accept that, but it could be done.

3. (clean-code) Remove/clean code

I've taken on fixing linting issues and removing some obvious dead code/comments. Also fixes some small issues I encountered.

4. (use-logging) Use of python logging module

Have the whole handling of debug.txt/log file handled by python and remove the piping of stdout/sterr from exec scripts

I still have more as of today:
- add some timer utils
- improve Qt imports
- fix tcx loader and start to add tests
but i guess it's enough already :) 

if you don't agree or don't have time, no hard feelings. Will just work on my fork

hishizuka, 2023-09-03:
Hi, @Ptosiek 
Thanks for fixing so many of the codes.
I am continuing to add features in a private repository and will release them soon.
If you can, I would appreciate any modifications you can make to that code. If not possible, I will check and merge them one by one.

BTW, to see the TCX sample do you live in Oregon? I'm using Rolf Prima wheels made in Oregon!

Ptosiek, 2023-09-05:
No problem.
I will wait for you to release what you have, and just rebase my branches and open PR then only. Just opening thw crdp one already.
If you can, ping me when you have release everything you wanted.

It's a coincidence, I just picked the first file I was shown on ridewithgps. I am living in Europe ;)

hishizuka, 2023-09-15:
Hi, @Ptosiek 
Finally, the repository has been updated. Your pull requests are welcome.

PR #15 (2023-08-28): use proper attribute
fix #16
commit 5ec93b98 (2023-08-28): use proper attribute
fix #16