hishizuka/pizero_bikecomputer #20: touchscreen compatibility

eliabruni, 2023-09-18:
Hello @hishizuka , thanks for this great project!

I am compiling a shopping list to try out your project and I was wondering what do you think about using this following screen: https://www.waveshare.com/5inch-DSI-LCD-B.htm? Do you think it would work as well?

Thank you!

hishizuka, 2023-09-19:
Hi, @eliabruni .
Thank you for using my program!

This screen would require a lot of power.

_Using a 5V power supply, the maximum brightness operating current is about 250mA and the minimum brightness operating current is about 150mA._
https://www.waveshare.com/wiki/5inch_DSI_LCD_(B)

Is there a place to attach a high-capacity mobile battery (10000mAh+~) ?

FYI: MIP display can reduce power consumption to less than 50mA

closed 2023-09-24 by hishizuka

eliabruni, 2023-09-25:
Hello @hishizuka , sorry for the late reply - thanks for the feedback! It's good to know about the MIP impact on power consumption. We are indeed considering to have a bike computer with quite more battery capacity as we are also including a cool AI inference engine into it. If you are interested, we can also chat about it in private.

hishizuka, 2023-09-25:
Hi @eliabruni ,
I am interested in your plan.
If you are going to use AI engine, Raspberry Pi zero is not powerful enough, you should at least use Raspberry Pi zero2/Radxa zero/Raspberry Pi CM4.
I would also like to have the AI engine in my program someday, can I just email you at your Gmail address?

eliabruni, 2023-09-26:
Hey @hishizuka, yes, feel free to drop a message to my gmail, thanks!