hishizuka/pizero_bikecomputer #46: I2C GPS

Ptosiek, 2023-10-12:
I know you wrote the support as 'experimental' but:
I've just got the Pimoroni breakout module and connected it though I2C (just because it's easier, and also wanted to test that code path)
The pa1010d lib does not return the track, and the "timestamp" field returned is a time object not a datetime.

But it's a lib limitation (cf. https://github.com/pimoroni/pa1010d-python/pull/12/files)

Should we:
1 Fork the lib
2 Reimplement it in this repo
3 Drop direct I2C support and somewhat pass it through GPSD anyway (cf https://ozzmaker.com/accessing-gps-via-i2c-on-a-berrygps-imu/), that suppose a bugger rewrite (mainly the way we detect/use the GPS sensor)

It's somewhat blocking this branch:
https://github.com/Ptosiek/pizero_bikecomputer/tree/gps
unless we decide to solve that later.
(The adafruit uart path is not tested either)

hishizuka, 2023-10-13:
Hi @Ptosiek ,
thank you for researching pa1010d-python Library.
Let's stop supporting I2C and adafruit UART.
I thought I2C connected GPS would be a bit more popular at the time. In the end, GPSd is good enough.

Ptosiek, 2023-10-13:
Could not make it work (i2c=> gpsd) just yet (will keep trying). I will prob just open the gps PR as is so you can have a look

closed 2024-07-16 by hishizuka