hishizuka/pizero_bikecomputer #48: log filename format

hishizuka, 2023-10-15:
Hi @Ptosiek ,

The log file name is determined by isoformat(), but ":" is a bit unwieldy because it requires an escape character.

We would like to change the format of the time as follows
What do you think?

- '20231015023506'
- '2023_10_15_02_35_06'
- '2023_10_15-02_35_06'
- '2023-10-15_02-35-06'

Ptosiek, 2023-10-15:
Hey,
Not sure what's the problem is ? When do you need to escape the log file name exactly?
(Using a standard format makes it easier to be parsed if needed, but if it only needs to be human readable, then the last option is def the better one)

hishizuka, 2023-10-15:
Hi @Ptosiek ,
between Linux and macOS, there is no problem. (Still, macOS finder shows "/" instead of ":".)
I often check logs on my cell phone while moving, but when I try to copy files to Android via FTP, I can't do it, so I have to rename them every time.

Ptosiek, 2023-10-16:
Let's change the format then

hishizuka, 2023-10-20:
Hi @Ptosiek ,
I modified log file format.

closed 2023-10-20 by hishizuka