hishizuka/pizero_bikecomputer #49: Error; coroutine awaited

maverick-2019jp, 2023-10-15:
Thank you very much for the wonderful project.
However, I have a problem starting the application and need help.
After following the guide to install this, and running it, I run into this error;

########## INITIALIZE START ##########
Raspberry Pi Zero W Rev 1.1(BCM2835), serial:0xd943328a
Initialize modules:
config import : 4.042 sec
config init   : 0.171 sec
display import: 0.022 sec
display init  : 0.000 sec
import gui    : 2.463 sec
import logger : 0.047 sec
total         : 6.745 sec
########## INITIALIZE END ##########
Traceback (most recent call last):
  File "/home/pi/pizero_bikecomputer/pizero_bikecomputer.py", line 56, in <module>
main()
  File "/home/pi/pizero_bikecomputer/pizero_bikecomputer.py", line 52, in main
gui_pyqt.GUI_PyQt(config)
  File "/home/pi/pizero_bikecomputer/modules/gui_pyqt.py", line 149, in __init__
self.init_window()
  File "/home/pi/pizero_bikecomputer/modules/gui_pyqt.py", line 155, in init_window
self.config.init_loop(call_from_gui=True)
  File "/home/pi/pizero_bikecomputer/modules/config.py", line 766, in init_loop
self.start_coroutine()
  File "/home/pi/pizero_bikecomputer/modules/config.py", line 771, in start_coroutine
self.logger.start_coroutine()
  File "/home/pi/pizero_bikecomputer/modules/logger_core.py", line 87, in start_coroutine
asyncio.create_task(self.sql_worker())
  File "/usr/lib/python3.9/asyncio/tasks.py", line 360, in create_task
loop = events.get_running_loop()
RuntimeError
:
no running event loop
sys:1: RuntimeWarning: coroutine 'LoggerCore.sql_worker' was never awaited

I'm running Pi OS 32 Bit (Bullseye Full) on Pi0 W with pi user

hishizuka, 2023-10-15:
Hi, @maverick-2019jp ,
thank you for using my program.

Sorry, this is a bug that has not been fixed yet.

If the version qasync is 0.24.2 or higher, it will not start. I will fix this issue soon.
The workaround is as follows.

```
sudo pip3 uninstall qasync
sudo pip3 install qasync==0.24.0
```

hishizuka, 2023-10-16:
Workaround for using the latest version of asynqt.

`init_loop` of modules/config.py

    def init_loop(self, call_from_gui=False):
        if self.G_GUI_MODE == "PyQt":
            if call_from_gui:
                asyncio.set_event_loop(self.loop)
                asyncio.events._set_running_loop(self.loop)  #add
                self.start_coroutine()
        else:

However, this calls an internal method of asyncqt, so the usage of eventloop needs to be revised.

maverick-2019jp, 2023-10-16:
Thank you for the information.
Operation confirmed with qasync version 0.24.0.

Thank you

closed 2023-10-25 by hishizuka