hishizuka/pizero_bikecomputer #51: I can't use MIP display ().

junk-junk, 2023-10-18:
I've trouble with MIP monitor. 
I could use MIP monitor by mip_test_pigpio with  a few changes (only clock speed).
however,When I run the pizero_bikecomputer, the screen lights up, but nothing is displayed (displays correctly in X-window).
[my setup guide in Japanese] says to use forked PyQtGraph, but I can't find it,Is this the cause?

[GENERAL]
display = MIP
OS:Raspberry pi OS bullseye
I installed qt5 according to https://forums.raspberrypi.com/viewtopic.php?t=346805

thank you.

hishizuka, 2023-10-18:
Hi, 

follow software documentation.
https://github.com/hishizuka/pizero_bikecomputer/blob/master/doc/software_installation.md#manual-execution

> displays correctly in X-window

Maybe the execution command is incorrect.

```
$ QT_QPA_PLATFORM=offscreen python3 pizero_bikecomputer.py
```

junk-junk, 2023-10-18:
Hi,hishizuka.
Thank you for comment

> $ QT_QPA_PLATFORM=offscreen python3 pizero_bikecomputer.py 

umm, window is disappered in x-window, but MIP display still blank(or displayed last image).

>If you don't mind, I would be glad to see pictures of the examples you have made.
> I have never seen anyone buy and build a MIP display before.

Of course!
but,It isn't assembled yet.
It will be equipped with an 18650 battery instead of Pi juice.I'll share it when it's finished!
[image]

junk-junk, 2023-10-18:
Hi, @hishizuka,
I think there no errors,so I wonder why it's not working.
Please let me know if there is anything else I can do.
Since it's indoors, GPS isn't working ;-(

pi@raspberrypi:~/pizero_bikecomputer $ python pizero_bikecomputer.py 
########## INITIALIZE START ##########
Raspberry Pi 3 Model B Plus Rev 1.3(BCM2835), serial:0xb63c9dc5
Initialize modules:
config import : 1.751 sec
config init   : 0.055 sec
display import: 0.011 sec
display init  : 3.140 sec
import gui    : 2.926 sec
import logger : 0.042 sec
total         : 7.925 sec
########## INITIALIZE END ##########
Qt version: 6.4.2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: mmh3 in /home/pi/.local/lib/python3.11/site-packages (4.0.1)
token is not set, connection without tls wont be established
detected sensor modules:
GPS
GPIO
I2C
detected I2c sensors:
[sensor] Initialize:
ANT+ : 0.001 sec
I2C  : 0.149 sec
total: 0.149 sec
[logger] Loading course:
read_file           : 0.001 sec
downsample          : 0.000 sec
calc_slope_smoothing: 0.000 sec
modify_course_points: 0.000 sec
total               : 0.001 sec
Drawing components:
misc  : 0.004 sec
import: 0.180 sec
init  : 1.927 sec
layout: 0.000 sec
total : 2.111 sec
delay init: 18.107 sec
try to modify time by gps...
['uname', '-v']
['sudo', 'date', '-u', '--set', '2023/10/18 15:12:34']
Wed Oct 18 03:12:34 PM UTC 2023

pi@raspberrypi:~/pizero_bikecomputer $ QT_QPA_PLATFORM=offscreen python3 pizero_bikecomputer.py
########## INITIALIZE START ##########
Raspberry Pi 3 Model B Plus Rev 1.3(BCM2835), serial:0xb63c9dc5
Initialize modules:
config import : 0.766 sec
config init   : 0.029 sec
display import: 0.004 sec
display init  : 1.935 sec
import gui    : 1.302 sec
import logger : 0.015 sec
total         : 4.051 sec
########## INITIALIZE END ##########
Qt version: 6.4.2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: mmh3 in /home/pi/.local/lib/python3.11/site-packages (4.0.1)
token is not set, connection without tls wont be established
detected sensor modules:
GPS
GPIO
I2C
detected I2c sensors:
[sensor] Initialize:
ANT+ : 0.000 sec
I2C  : 0.116 sec
total: 0.116 sec
[logger] Loading course:
read_file           : 0.000 sec
downsample          : 0.000 sec
calc_slope_smoothing: 0.000 sec
modify_course_points: 0.000 sec
total               : 0.000 sec
Drawing components:
misc  : 0.003 sec
import: 0.095 sec
init  : 1.609 sec
layout: 0.000 sec
total : 1.707 sec
delay init: 14.414 sec
try to modify time by gps...
['uname', '-v']
['sudo', 'date', '-u', '--set', '2023/10/18 15:16:45']
Wed Oct 18 03:16:45 PM UTC 2023

hishizuka, 2023-10-18:
Sorry,
I made a fatal mistake. Pin assign of SCS was changed from GPIO 22(15 pin) to GPIO 23(16 pin).

https://github.com/hishizuka/pizero_bikecomputer/blob/master/doc/hardware_installation.md#displays-without-phat-headers-mip-reflective-color-lcd-and-sharp-memory-display

junk-junk, 2023-10-20:
Hi,
sorry,late reply. I was explaining my credit card statement to my wife (especially about the sushi restaurant).

>Qt version is 6.4.2, so it must be bookwarm?

Yes,I'm using bookworm. So, I tried two method.

1. One: I installed python3-pyqt6

>   python3-pyqtgraph is already the newest version (0.13.1-4).
python3-pyqt6 is already the newest version (6.4.2-1+rpi1).
libqt6svg6 is already the newest version (6.4.2-2).

2. I installed the old version (sudo apt build - dep python three - pyqtgraph=0.11.1-1).

However, there was no change. The window opens in the X-window, but it is not displayed in the MIP display in both case.

>pi@raspberrypi:~/pizero_bikecomputer $ ./pizero_bikecomputer.py --demo
########## INITIALIZE START ##########
Raspberry Pi 3 Model B Plus Rev 1.3(BCM2835), serial:0xb63c9dc5
MIP DISPLAY: True
Initialize modules:
config import : 0.745 sec
config init   : 0.028 sec
display import: 0.004 sec
display init  : 1.882 sec
import gui    : 1.104 sec
import logger : 0.014 sec
total         : 3.777 sec
########## INITIALIZE END ##########
Qt version: 5.15.8
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: mmh3 in /home/pi/.local/lib/python3.11/site-packages (4.0.1)
token is not set, connection without tls wont be established
detected sensor modules:
GPS
GPSD chosen
GPIO
I2C
detected I2c sensors:
[sensor] Initialize:
ANT+ : 0.000 sec
I2C  : 0.076 sec
total: 0.076 sec
[logger] Loading course:
read_file           : 0.000 sec
downsample          : 0.000 sec
calc_slope_smoothing: 0.000 sec
modify_course_points: 0.000 sec
total               : 0.001 sec
Drawing components:
misc  : 0.002 sec
import: 0.073 sec
init  : 1.471 sec
layout: 0.000 sec
total : 1.546 sec
delay init: 15.308 sec
try to modify time by gps to 2023-10-20T14:41:04.970Z
Fri 20 Oct 14:41:04 UTC 2023

hishizuka, 2023-10-20:
Hi @junk-junk ,
I ate sushi yesterday, too.

First, did the latest mip_test_pigpio.py work? (Updated yesterday.) 
If Yes, then it is a matter of installation or activation method.

Whenever you use MIP display, you must activate it with `QT_QPA_PLATFORM=offscreen`.
Without it, the logic for flowing data to the SPI will not work.

```
QT_QPA_PLATFORM=offscreen python3 pizero_bikecomputer.py
```

If this still does not work, then it is an installation problem.
PyQtGraph is OK with the latest version.

junk-junk, 2023-10-21:
Hi,@hishizuka
Thanks you for your great help.
Sorry,,,I had a big mistake....

>First, did the latest mip_test_pigpio.py work? (Updated yesterday.)

It doesn't work, and I realized my miswiring.
I installed the hardware according to the py_mip_sample and Japasese document.
But, in pizero_bikecomputer, CS is changed from IO15 to IO16. So, I rewired the cable and it worked!
Thank you!
Best regard!

hishizuka, 2023-10-21:
Hi @junk-junk ,
that was good.
Enjoy!!!

closed 2023-10-21 by hishizuka