hishizuka/pizero_bikecomputer #59: ant usb not working on RPiZeroW or RPiZero2W

pbljung, 2023-11-01:
Hi, I'm trying to run openant or bikecomputer on a RPi. I can successfully run bikecomputer on a mac where it finds an Ant+ heartrate sensor and displays the HR. 

On a RPi 3B v1.2 I can successfully run the openant examples to see the HR using an ANT+ usb dongle.

But if I move the sdcard and usb to a RPiZero2W then the usb is never found. Dmesg shows 
```
[   44.984439] usb 1-1: new full-speed USB device number 2 using dwc2
[   45.204440] usb 1-1: device descriptor read/64, error -71
[   45.544545] usb 1-1: device descriptor read/64, error -71
[   45.884446] usb 1-1: new full-speed USB device number 3 using dwc2
[   46.104446] usb 1-1: device descriptor read/64, error -71
[   46.444438] usb 1-1: device descriptor read/64, error -71
[   46.564514] usb usb1-port1: attempt power cycle
```
Supposedly this error has to do with inadequate power. However the power sources should be adequate (either mac powerbook (usbC port) or my powerbook charger (usbC port)) for all testing of RPi 3B, ZeroW and Zero2W. There is nothing attached to the RPi computers other than the ant usb dongle. I have 2 dongles (marked "Garmin" or "ANT+") and both work fine on my mac. 

If I move the sdcard to a new 2nd RpiZero2W then it fails with the same issue.

If I move the sdcard to a RPiZeroW then the usb is recognized (dmesg, lsusb, /dev/ttyUSB0), but the openant examples fail e.g. 
```
python3 scan.py 
Driver available: [<class 'ant.base.driver.USB2Driver'>, <class 'ant.base.driver.USB3Driver'>]
 - Using: <class 'ant.base.driver.USB3Driver'>
Traceback (most recent call last):
  File "scan.py", line 78, in <module>
    main()
  File "scan.py", line 55, in main
    node = Node()
  File "/home/pi/Documents/ant/openant/ant/easy/node.py", line 55, in __init__
    self.ant = Ant()
  File "/home/pi/Documents/ant/openant/ant/base/ant.py", line 60, in __init__
    self._driver.open()
  File "/home/pi/Documents/ant/openant/ant/base/driver.py", line 202, in open
    dev.set_configuration()
  File "/home/pi/.local/lib/python3.7/site-packages/usb/core.py", line 915, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/home/pi/.local/lib/python3.7/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/usb/core.py", line 159, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "/home/pi/.local/lib/python3.7/site-packages/usb/backend/libusb1.py", line 812, in set_configuration
    _check(self.lib.libusb_set_configuration(dev_handle.handle, config_value))
  File "/home/pi/.local/lib/python3.7/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno None] Other error
```
How do I get openant running on a RpiZero or RpiZero2? 
Thanks

hishizuka, 2023-11-02:
Hi, @pbljung ,

I am sorry, but I cannot imagine your situation.

pbljung, 2023-11-02:
The ANT+ usb dongle doesn't fit the ports of a RPiZero, so I plug a micro-usb cable adapter into the OTG port, and a Dynastream 1009 ANT+ USB dongle into the adapter. Power is from my mac powerbook (or from my mac powerbook 140W charger) into the RPi power microusb port. No additional hw attached to the RPi. No usb hubs are used. RPi sdcard is older (buster) but "plain jane" (i2c enabled, spi enabled, serial console disabled), and not programmed to use any special OTG capabilities.

hishizuka, 2023-11-02:
> a micro-usb cable adapter

Isn't the difference between RPi3 and RPi Zero the USB adapter?
Is it a proper USB OTG adapter?
Can you try to connect a USB keyboard with your micro-usb cable adapter and HDMI cable to the RPi zero and operate the console?

https://github.com/hishizuka/pizero_bikecomputer_v1/blob/master/doc/hardware_installation.md#ant-usb-dongle

pbljung, 2023-11-02:
I saw that  bullseye is the earliest OS for Rpizero2W so I created a new sdcard with 2023-05-03-raspios-bullseye-armhf-lite.img  Running on RpizeroW and Rpizero2W now dmesg and lsusb both return expected results (ie the ant dongle is recognized). 

I am using this USB OTG adapter https://www.adafruit.com/product/1099

Now examples run fine on RpizeroW
```
pi@raspberrypi:~/Documents/ant/openant $ python3 examples/scan.py 
Driver available: [<class 'ant.base.driver.SerialDriver'>, <class 'ant.base.driver.USB2Driver'>, <class 'ant.base.driver.USB3Driver'>]
 - Using: <class 'ant.base.driver.USB3Driver'>
array('B', [4, 0, 228, 209, 95, 213, 4, 71, 128, 172, 30, 120, 1])
New Device Found: 7852 of type 120
array('B', [132, 0, 228, 209, 95, 213, 4, 71, 128, 172, 30, 120, 1])
New Device Found: 7852 of type 120
array('B', [132, 0, 95, 213, 214, 216, 5, 71, 128, 172, 30, 120, 1])
New Device Found: 7852 of type 120
array('B', [4, 0, 95, 213, 214, 216, 5, 71, 128, 172, 30, 120, 1])
New Device Found: 7852 of type 120
```
```
pi@raspberrypi:~/Documents/ant/openant $ python3 examples/heart_rate_monitor.py 
Driver available: [<class 'ant.base.driver.SerialDriver'>, <class 'ant.base.driver.USB2Driver'>, <class 'ant.base.driver.USB3Driver'>]
 - Using: <class 'ant.base.driver.USB3Driver'>
<class 'usb.core.USBTimeoutError'>, (110, 'Operation timed out')
Heartrate: 63 [BPM]
```

However on the Rpizero2W the openant examples still fail e.g. python3 examples/scan.py
```
Driver available: [<class 'ant.base.driver.SerialDriver'>, <class 'ant.base.driver.USB2Driver'>, <class 'ant.base.driver.USB3Driver'>]
 - Using: <class 'ant.base.driver.USB3Driver'>
Traceback (most recent call last):
  File "/home/pi/Documents/ant/openant/examples/scan.py", line 78, in <module>
    main()
  File "/home/pi/Documents/ant/openant/examples/scan.py", line 55, in main
    node = Node()
  File "/usr/local/lib/python3.9/dist-packages/ant/easy/node.py", line 55, in __init__
    self.ant = Ant()
  File "/usr/local/lib/python3.9/dist-packages/ant/base/ant.py", line 60, in __init__
    self._driver.open()
  File "/usr/local/lib/python3.9/dist-packages/ant/base/driver.py", line 202, in open
    dev.set_configuration()
  File "/usr/local/lib/python3.9/dist-packages/usb/core.py", line 915, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/usr/local/lib/python3.9/dist-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/usb/core.py", line 159, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "/usr/local/lib/python3.9/dist-packages/usb/backend/libusb1.py", line 812, in set_configuration
    _check(self.lib.libusb_set_configuration(dev_handle.handle, config_value))
  File "/usr/local/lib/python3.9/dist-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno None] Other error
```
Have you tried openant on Rpizero2W? If so which OS version? Which version of pyusb or libusb?

hishizuka, 2023-11-02:
I mainly use Raspberry Pi OS bullseye. 

I use libusb and python3-usb, as described in the document.

```
$ sudo apt-get install libusb-1.0-0 python3-usb
$ sudo pip3 install git+https://github.com/hishizuka/openant.git
```

Did you successfully execute pizero_bikecomputer.py with Raspberry Pi zero?

hishizuka, 2023-11-02:
BTW, 
Why don't you try this one? I use this combination.

https://www.adafruit.com/product/2910
https://ja.aliexpress.com/item/1005001355302356.html?spm=a2g0o.detail.0.0.3a904ff3U9iMFH&gps-id=pcDetailTopMoreOtherSeller&scm=1007.40050.362094.0&scm_id=1007.40050.362094.0&scm-url=1007.40050.362094.0&pvid=1569c3aa-928f-4f63-b961-2ebcd9091824&_t=gps-id:pcDetailTopMoreOtherSeller,scm-url:1007.40050.362094.0,pvid:1569c3aa-928f-4f63-b961-2ebcd9091824,tpp_buckets:668%232846%238111%231996&isseo=y&pdp_npi=4%40dis%21JPY%211095%21612.0%21%21%217.15%21%21%402141113616989611252423464e1314%2112000026755529768%21rec%21JP%21%21AB

[image]

closed 2023-11-08 by hishizuka

pbljung, 2023-11-12:
Apparently all microusb to usb A adapters are not alike!

This adapter cable works on RpiZeroW but not RpiZero2W
https://www.adafruit.com/product/898

This "otg adapter" works on RpiZero2W
https://www.adafruit.com/product/2910

My "authentic" ANT+ dongles don't have the bottom exterior metal casing
which the micro otg adapter presses against, so I ziptied them together.
A RPiZero2W running Bullseye now sees the (authentic) ANT+ dongle (dmesg,
lsusb, openant examples).