hishizuka/pizero_bikecomputer #62: Google direction API doesn't work

joeydejong112, 2023-11-08:
Hi, i worked with your application, but i cant seem to get the Google direction api working

First of all 
I got the API key from google direction api.
I putted the key into the google direction section in settings.conf but this doesn't seem to work it keeps disappearing, and also i don't get the G option in maps to search for any routes.

joeydejong112, 2023-11-08:
I got the Go button working( the key stays now) but it just doesn't do anything when i click on it

hishizuka, 2023-11-08:
It is still a confusing GUI.
After unlocking the map movement lock with the "L" key on the left, use the arrow keys on the right to move to the destination. Then press the "G" button. The route from the initial position to the center position is calculated.

BTW, this function is difficult to use with hardware buttons, so we are creating a function to transfer routes created with Google Maps on Android.

https://github.com/hishizuka/pizero_bikecomputer/tree/master/doc#courses
"Android Google Maps"

closed 2023-11-08 by hishizuka