hishizuka/pizero_bikecomputer #66: frequent timeouts when starting

pbljung, 2023-11-12:
Using the OTG adapter https://www.adafruit.com/product/2910 and an authentic ANT+ dongle I got Openant to run on a RPiZero2W on Bullseye. The cable adapter https://www.adafruit.com/product/898 works on RpiZeroW but not RpiZero2W.

I want to get stable ANT+ data before trying to run pizero_bikecomputer. However I get frequent AntExceptions and timeouts when running openant examples e.g. heart_rate_monitor.py. I ran heart_rate_monitor 20 times and 5 runs run fail like this. There appear to be 2 issues: a) Setting network key sometimes fails, and b) USBError times out (this error prints at 1Hz like the heartbeat, but no heartbeat is printed). 

Sometimes it runs, sometimes not. I get similar errors running other openant examples.
Any suggestions?

> pi@raspberrypi:~/Documents/ant $ python3 heart_rate_monitor.py
> Driver available: [\<class ant.base.driver.SerialDriver>, \<class ant.base.driver.USB2Driver>, \<class ant.base.driver.USB3Driver>]
> Using: \<class ant.base.driver.USB3Driver>
> \<class usb.core.USBError>, (110, Operation timed out)
> Heartrate: 82 [BPM]
> Heartrate: 82 [BPM]
> Heartrate: 82 [BPM]
> Heartrate: 81 [BPM]
> 

> pi@raspberrypi:~/Documents/ant $ python3 heart_rate_monitor.py
> Driver available: [\<class ant.base.driver.SerialDriver>, \<class ant.base.driver.USB2Driver>, \<class ant.base.driver.USB3Driver>]
> Using: <class ant.base.driver.USB3Driver>
> \<class usb.core.USBError>, (110, Operation timed out)
> \<class usb.core.USBError>, (110, Operation timed out)
> \<class usb.core.USBError>, (110, Operation timed out)
> \<class usb.core.USBError>, (110, Operation timed out)
> \<class usb.core.USBError>, (110, Operation timed out)
> \<class usb.core.USBError>, (110, Operation timed out)
> \<class usb.core.USBError>, (110, Operation timed out)
> \<class usb.core.USBError>, (110, Operation timed out)
> \<class usb.core.USBError>, (110, Operation timed out)
> \<class usb.core.USBError>, (110, Operation timed out)
> Traceback (most recent call last):
>   File "/home/pi/Documents/ant/heart_rate_monitor.py", line 69, in <module>
>     main()
>   File "/home/pi/Documents/ant/heart_rate_monitor.py", line 49, in main
>     node.set_network_key(0x00, NETWORK_KEY)
>   File "/usr/local/lib/python3.9/dist-packages/ant/easy/node.py", line 84, in set_network_key
>     return self.wait_for_response(Message.ID.SET_NETWORK_KEY)
>   File "/usr/local/lib/python3.9/dist-packages/ant/easy/node.py", line 103, in wait_for_response
>     return wait_for_response(event_id, self._responses, self._responses_cond)
>   File "/usr/local/lib/python3.9/dist-packages/ant/easy/filter.py", line 98, in wait_for_response
>     return wait_for_message(match, process, queue, condition)
>   File "/usr/local/lib/python3.9/dist-packages/ant/easy/filter.py", line 62, in wait_for_message
>     raise AntException("Timed out while waiting for message")
> ant.easy.exception.AntException: Timed out while waiting for message
> \<class usb.core.USBError>, (110, Operation timed out)
> \<class usb.core.USBError>, (110, Operation timed out)
>

hishizuka, 2023-11-12:
The node.stop() must be run at the exit. Otherwise, it will time out one out of every two times.
That sample program does not move node.stop() because it suddenly KILLS with CTRL-C.

No need to stay with the sample program.

closed 2023-11-12 by hishizuka

pbljung, 2023-11-13:
I saw that your rpizero_bikecomputer ANT code is much more similar to
Tigge's examples than your own examples.
I found that adding channel.close() to a finally block results in examples
that run EVERY time.

    try:
>       channel.open()
>       channel_cadence_speed.open()
>       node.start()
>     except KeyboardInterrupt:
>       print("Pressed CtrlC, shutting down")
>     finally:
>       print("FINALLY, closing ANT+")
>       channel.close()
>       channel_cadence_speed.close()
>       node.stop()

You should update your examples!

/Per