hishizuka/pizero_bikecomputer #71: Radar support via LD2450?

Fmstrat, 2024-05-01:
So I've been building ESPHome sensors using an LD2450. These are purpose built for detecting human presence, but could something similar be done to use as a rear radar similar to a Garmin Varia?

Theoretically it could be wired via GPIO, or with a ESP mini and a small battery wirelessly. The LD2450 also has Bluetooth built in, but I'm not sure of an interface method with the Pi0, or if this would even be a good sensor.

hishizuka, 2024-05-01:
According to the specifications for LD2450, the maximum detection distance is 6 m, making it a useless alternative to Garmin Varia.
If the difference in speed is 20 km/h, this distance can be caught up in 1 second.

closed 2024-05-01 by hishizuka

Fmstrat, 2024-05-01:
Well yes, but that's antenna design. Both the Terasilic and the Garmin Varia use a 24ghz sensor, just like the LD2450, but configured for lower resolution at a 140 meter range. There are plenty of other sensors on the market with longer range.

hishizuka, 2024-05-01:
Are there readily available sensors on the market with longer range?

Fmstrat, 2024-05-01:
A quick search found this from the same manufacturer as the LD2450: https://www.aliexpress.us/item/3256806493734631.html

No idea of it's any good, but it's cheap and everywhere.